bmax121/APatch #1028: “安装到未使用的槽位”后重启变砖

huangsijun17, 2025-04-19:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

一加11使用“安装到未使用的槽位”后重启变砖

### Reproduce method/复现方法

- 1. 安装ColorOS PHB110_15.0.0.600(CN01) F.25
- 2. 在系统更新更新系统
- 3. 在Apatch里点击安装到未使用的槽位

### Expected behavior/预期行为

正常启动

### Actual behaviour /实际行为

重启出现第一屏幕logo后黑屏。接电脑显示进入900E模式

### Device Name/设备名称

一加11

### OS Version/系统版本

ColorOS PHB110_15.0.0.600(CN01) F.25升级ColorOS PHB110_15.0.0.700(CN01) F.26

### APatch Version/APatch 版本

找不出来了，应该是最新RELEASE

### Kernel Version/内核版本

找不出来了

### KernelPatch Version/KernelPatch 版本

找不出来了，应该是最新RELEASE

### Additional context/其他信息

文档里写了：
> APatch 的 OTA 功能目前不太稳定，可能会出现问题。当遇到问题时，最好前往 APatch 项目 GitHub 仓库提交 issue。

所以提的。（个人感觉不想是 APatch 的问题）

1a2b3clong, 2025-04-24:
我也是，想问怎么解决了呢，一个卡槽开机界面不断重启，一个卡槽开机界面之后就黑屏了，刷了原boot也没恢复呢

1a2b3clong, 2025-04-24:
> > 我也是，想问怎么解决了呢，一个卡槽开机界面不断重启，一个卡槽开机界面之后就黑屏了，刷了原boot也没恢复呢
> 
> 今天去了趟售后。（哭

谢谢你告诉我这个消息😭

1a2b3clong, 2025-04-24:
但是吧，我又重新刷了一遍vbmeta 和vbmeta_system   --disable-verity  --disable-verification  然后就又可以咯

huangsijun17, 2025-04-24:
> 但是吧，我又重新刷了一遍vbmeta 和vbmeta_system --disable-verity --disable-verification 然后就又可以咯

我刷了不行。

zyklone4096, 2025-05-15:
我有类似的情况，槽位A重启黑屏，槽位B正常，测试过11039和11071

<details>
<summary>修补日志</summary>

```
****************************
****************************
 APatch Boot Image Patcher
****************************
- Patching kernel
+ ./kptools -p -i kernel.ori -S Jiji0808@ -k kpimg -o kernel true
[+] kernel image_size: 0x02ba6190
[+] kernel uefi header: true
[+] kernel load_offset: 0x00000000
[+] kernel kernel_size: 0x02d00000
[+] kernel page_shift: 12
[+] patched kernel image ...
[+] linux_banner 1: Linux version 5.15.176-gfd0e1f2272f8 (build-user@build-host) (Android (12701618, +pgo, +bolt, +lto, +mlgo, based on r536225) clang version 19.0.1 (https://android.googlesource.com/toolchain/llvm-project b3a530ec6537146650e42be89f1089e9a3588460), LLD 19.0.1) #1 SMP PREEMPT Wed May 7 08:59:59 UTC 2025
[+] linux_banner offset: 0x1c155d2
[+] kernel version major: 5, minor: 15, patch: 176
[+] kallsyms_token_table offset: 0x01b1c920
[+] endian: little
[+] kallsyms_token_index offset: 0x01b1cc78
[+] arm64 relocation kernel_va: 0xffffffffffffffff
[?] can't find arm64 relocation table
[+] kallsyms_markers range: [0x01b1bcd0, 0x01b1c91c), count: 0x00000312
[+] approximate kallsyms_offsets range: [0x0164d0d4, 0x01711b74) count: 0x000312a8
[+] kallsyms_names offset: 0x01711b80
[?] can't find kallsyms_num_syms, try: 0x0003129e
[+] names table linux_banner index: 0x0001ce1f
[+] linux_banner index: 0
[+] kallsyms_offsets offset: 0x0164d0d8
[+] pid_vnr: type: T, offset: 0x0018212c
[+] pid_vnr verfied sp_el0, insn: 0xd5384108
[+] layout kimg: 0x0,0x2b79a00, kpimg: 0x2b7a000,0x2c110, extra: 0x2ba6110,0x80, end: 0x2ba6190, start: 0x2d00000
[+] kpimg version: b02
[+] kpimg compile time: 01:05:21 Feb 12 2025
[+] kpimg config: android, release
[+] tcp_init_sock: type: T, offset: 0x00fae9c8
[+] map_start: 0xfae9d0, max_size: 0x800
[+] kallsyms_lookup_name: type: T, offset: 0x00251524
[?] no symbol: printk
[+] _printk: type: T, offset: 0x001e4164
[+] memblock_reserve: type: T, offset: 0x003f2554
[+] memblock_free: type: T, offset: 0x003f23a4
[+] memblock_mark_nomap: type: T, offset: 0x003f2890
[+] memblock_phys_alloc_try_nid: type: T, offset: 0x024f6ed0
[?] no symbol: memblock_virt_alloc_try_nid
[+] memblock_alloc_try_nid: type: T, offset: 0x024f71b4
[+] panic: type: T, offset: 0x00151070
[+] rest_init: type: T, offset: 0x0120dab0
[+] kernel_init: type: t, offset: 0x0120db98
[?] no symbol: report_cfi_failure
[+] __cfi_slowpath_diag: type: T, offset: 0x0033f268
[?] no symbol: __cfi_slowpath
[+] copy_process: type: t, offset: 0x0014caa0
[+] avc_denied: type: t, offset: 0x006bdaa8
[+] slow_avc_audit: type: T, offset: 0x006bc744
[+] input_handle_event: type: t, offset: 0x00bbc6a0
[+] root superkey hash: b2b9b8b52dc2c948a642f1170b92e7dea5ecb17a9f8dd2cbca0db899063ca0fb
[+] paging_init: type: T, offset: 0x024e1fd8
[+] patch done: kernel
+ patch_rc=0
+ set +x
- Repacking boot image
- Flashing new boot image
- Successfully Flashed!
- Connecting boot hal...
Error getting bootctrl v1.0 module.
[X] Failed to connect to boot hal, you may need switch slot manually
- Reboot to finish the installation...
****************************
```
</details>

zyklone4096, 2025-05-16:
我有类似的情况，槽位A重启黑屏，槽位B正常，测试过11039和11071

<details>
<summary>修补日志</summary>

```
****************************
****************************
 APatch Boot Image Patcher
****************************
- Patching kernel
+ ./kptools -p -i kernel.ori -S **** -k kpimg -o kernel true
[+] kernel image_size: 0x02ba6190
[+] kernel uefi header: true
[+] kernel load_offset: 0x00000000
[+] kernel kernel_size: 0x02d00000
[+] kernel page_shift: 12
[+] patched kernel image ...
[+] linux_banner 1: Linux version 5.15.176-gfd0e1f2272f8 (build-user@build-host) (Android (12701618, +pgo, +bolt, +lto, +mlgo, based on r536225) clang version 19.0.1 (https://android.googlesource.com/toolchain/llvm-project b3a530ec6537146650e42be89f1089e9a3588460), LLD 19.0.1) #1 SMP PREEMPT Wed May 7 08:59:59 UTC 2025
[+] linux_banner offset: 0x1c155d2
[+] kernel version major: 5, minor: 15, patch: 176
[+] kallsyms_token_table offset: 0x01b1c920
[+] endian: little
[+] kallsyms_token_index offset: 0x01b1cc78
[+] arm64 relocation kernel_va: 0xffffffffffffffff
[?] can't find arm64 relocation table
[+] kallsyms_markers range: [0x01b1bcd0, 0x01b1c91c), count: 0x00000312
[+] approximate kallsyms_offsets range: [0x0164d0d4, 0x01711b74) count: 0x000312a8
[+] kallsyms_names offset: 0x01711b80
[?] can't find kallsyms_num_syms, try: 0x0003129e
[+] names table linux_banner index: 0x0001ce1f
[+] linux_banner index: 0
[+] kallsyms_offsets offset: 0x0164d0d8
[+] pid_vnr: type: T, offset: 0x0018212c
[+] pid_vnr verfied sp_el0, insn: 0xd5384108
[+] layout kimg: 0x0,0x2b79a00, kpimg: 0x2b7a000,0x2c110, extra: 0x2ba6110,0x80, end: 0x2ba6190, start: 0x2d00000
[+] kpimg version: b02
[+] kpimg compile time: 01:05:21 Feb 12 2025
[+] kpimg config: android, release
[+] tcp_init_sock: type: T, offset: 0x00fae9c8
[+] map_start: 0xfae9d0, max_size: 0x800
[+] kallsyms_lookup_name: type: T, offset: 0x00251524
[?] no symbol: printk
[+] _printk: type: T, offset: 0x001e4164
[+] memblock_reserve: type: T, offset: 0x003f2554
[+] memblock_free: type: T, offset: 0x003f23a4
[+] memblock_mark_nomap: type: T, offset: 0x003f2890
[+] memblock_phys_alloc_try_nid: type: T, offset: 0x024f6ed0
[?] no symbol: memblock_virt_alloc_try_nid
[+] memblock_alloc_try_nid: type: T, offset: 0x024f71b4
[+] panic: type: T, offset: 0x00151070
[+] rest_init: type: T, offset: 0x0120dab0
[+] kernel_init: type: t, offset: 0x0120db98
[?] no symbol: report_cfi_failure
[+] __cfi_slowpath_diag: type: T, offset: 0x0033f268
[?] no symbol: __cfi_slowpath
[+] copy_process: type: t, offset: 0x0014caa0
[+] avc_denied: type: t, offset: 0x006bdaa8
[+] slow_avc_audit: type: T, offset: 0x006bc744
[+] input_handle_event: type: t, offset: 0x00bbc6a0
[+] root superkey hash: b2b9b8b52dc2c948a642f1170b92e7dea5ecb17a9f8dd2cbca0db899063ca0fb
[+] paging_init: type: T, offset: 0x024e1fd8
[+] patch done: kernel
+ patch_rc=0
+ set +x
- Repacking boot image
- Flashing new boot image
- Successfully Flashed!
- Connecting boot hal...
Error getting bootctrl v1.0 module.
[X] Failed to connect to boot hal, you may need switch slot manually
- Reboot to finish the installation...
****************************
```
</details>

更新ROM后才出问题，不确定是哪边导致的

so1ve, 2025-08-14:
> 我有类似的情况，槽位A重启黑屏，槽位B正常，测试过11039和11071
> 
> 修补日志
> ```
> ****************************
> ****************************
>  APatch Boot Image Patcher
> ****************************
> - Patching kernel
> + ./kptools -p -i kernel.ori -S **** -k kpimg -o kernel true
> [+] kernel image_size: 0x02ba6190
> [+] kernel uefi header: true
> [+] kernel load_offset: 0x00000000
> [+] kernel kernel_size: 0x02d00000
> [+] kernel page_shift: 12
> [+] patched kernel image ...
> [+] linux_banner 1: Linux version 5.15.176-gfd0e1f2272f8 (build-user@build-host) (Android (12701618, +pgo, +bolt, +lto, +mlgo, based on r536225) clang version 19.0.1 (https://android.googlesource.com/toolchain/llvm-project b3a530ec6537146650e42be89f1089e9a3588460), LLD 19.0.1) #1 SMP PREEMPT Wed May 7 08:59:59 UTC 2025
> [+] linux_banner offset: 0x1c155d2
> [+] kernel version major: 5, minor: 15, patch: 176
> [+] kallsyms_token_table offset: 0x01b1c920
> [+] endian: little
> [+] kallsyms_token_index offset: 0x01b1cc78
> [+] arm64 relocation kernel_va: 0xffffffffffffffff
> [?] can't find arm64 relocation table
> [+] kallsyms_markers range: [0x01b1bcd0, 0x01b1c91c), count: 0x00000312
> [+] approximate kallsyms_offsets range: [0x0164d0d4, 0x01711b74) count: 0x000312a8
> [+] kallsyms_names offset: 0x01711b80
> [?] can't find kallsyms_num_syms, try: 0x0003129e
> [+] names table linux_banner index: 0x0001ce1f
> [+] linux_banner index: 0
> [+] kallsyms_offsets offset: 0x0164d0d8
> [+] pid_vnr: type: T, offset: 0x0018212c
> [+] pid_vnr verfied sp_el0, insn: 0xd5384108
> [+] layout kimg: 0x0,0x2b79a00, kpimg: 0x2b7a000,0x2c110, extra: 0x2ba6110,0x80, end: 0x2ba6190, start: 0x2d00000
> [+] kpimg version: b02
> [+] kpimg compile time: 01:05:21 Feb 12 2025
> [+] kpimg config: android, release
> [+] tcp_init_sock: type: T, offset: 0x00fae9c8
> [+] map_start: 0xfae9d0, max_size: 0x800
> [+] kallsyms_lookup_name: type: T, offset: 0x00251524
> [?] no symbol: printk
> [+] _printk: type: T, offset: 0x001e4164
> [+] memblock_reserve: type: T, offset: 0x003f2554
> [+] memblock_free: type: T, offset: 0x003f23a4
> [+] memblock_mark_nomap: type: T, offset: 0x003f2890
> [+] memblock_phys_alloc_try_nid: type: T, offset: 0x024f6ed0
> [?] no symbol: memblock_virt_alloc_try_nid
> [+] memblock_alloc_try_nid: type: T, offset: 0x024f71b4
> [+] panic: type: T, offset: 0x00151070
> [+] rest_init: type: T, offset: 0x0120dab0
> [+] kernel_init: type: t, offset: 0x0120db98
> [?] no symbol: report_cfi_failure
> [+] __cfi_slowpath_diag: type: T, offset: 0x0033f268
> [?] no symbol: __cfi_slowpath
> [+] copy_process: type: t, offset: 0x0014caa0
> [+] avc_denied: type: t, offset: 0x006bdaa8
> [+] slow_avc_audit: type: T, offset: 0x006bc744
> [+] input_handle_event: type: t, offset: 0x00bbc6a0
> [+] root superkey hash: b2b9b8b52dc2c948a642f1170b92e7dea5ecb17a9f8dd2cbca0db899063ca0fb
> [+] paging_init: type: T, offset: 0x024e1fd8
> [+] patch done: kernel
> + patch_rc=0
> + set +x
> - Repacking boot image
> - Flashing new boot image
> - Successfully Flashed!
> - Connecting boot hal...
> Error getting bootctrl v1.0 module.
> [X] Failed to connect to boot hal, you may need switch slot manually
> - Reboot to finish the installation...
> ****************************
> ```
> 
> 更新ROM后才出问题，不确定是哪边导致的
> 
> 更新：重新sideload和fastboot flash双槽位后修复，但槽位B无法使用人脸解锁，使用11071

你也有`Error getting bootctrl v1.0 module.`的问题？我也是这样，在apatch管理器内使用ota模式修补后报了这个错，导致无法切换插槽，不知道是系统还是apatch的问题

commit 4453cf34 (2025-07-07): Revert "fix resymilink access error"

Close #1028.

wtf is that, completely made bootctl broken while fixed nothing.

This reverts commit c3e2b0296958e440e183592f6a55ddb0eb2b2134.