bmax121/APatch #1029: Detecting Root via /system/bin RW Mount on Reboot (Apatch 0.11.2)

zuluman, 2025-04-19:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

I am using Apatch version 0.11.2. Some applications check if the device is rooted by inspecting the /system/bin directory and detecting if it is mounted as read-write (rw). On my device, the /system/bin directory is mounted as rw after every reboot, which causes these applications to detect the device as rooted.

### Reproduce method/复现方法

.

### Expected behavior/预期行为

 /system/bin directory is mounted as read-write (rw) after each reboot

### Actual behaviour /实际行为

....

### Screenshots/截图

[image]
[image]

### Device Name/设备名称

mi 13 fuxi

### OS Version/系统版本

15 AQ3A.240912.001 |

### APatch Version/APatch 版本

0.11.2

### Kernel Version/内核版本

5.15.148-android13-8-00004-ge488687c12ef-ab11 838684

### KernelPatch Version/KernelPatch 版本

0.11.2

### Additional context/其他信息

 /system/bin directory is mounted as read-write (rw) after each reboot

salpikaw, 2025-04-30:
The very same issue I'm having. I hope a fix comes soon

imgoodgamerbro, 2025-05-27:
Same for me also

strec007, 2025-09-03:
I have the same issue with v0.12.0. Has anyone find out where the problem is? I disabled all modules but /system was still rw.

strec007, 2025-09-24:
> > I have the same issue with v0.12.0. Has anyone find out where the problem is? I disabled all modules but /system was still rw.
> > 
> > Edit: RootBeer does not detect RW Paths if Lie Mode is switched on in Apatch.
> 
> Where is Lie Mode ?
Sorry, I meant Lite Mode.