bmax121/APatch #1088: piping command into su not working like in other su varaints (echo whoami | su)

hg42, 2025-06-13:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

I am/was a developer/contributor of Neo Backup.
At some point in the past, we already had this problem and it was fixed, but since lately it is back again.

`echo some_command | su`
should execute `some_command` as root and not echo the command.

e.g.
`echo whoami | su`
outputs "whoami", instead of the result "root"

I know, that this works:
`echo whoami | su --no-pty`

However, this option isn't known by any other su.

I think `su` should automatically enable `--no-pty` if there is not tty.

Several apps and desktop programs use adb with piping commands into su.
I think this is the only way, that su works reliably, because escaping/quoting is handled quite differently by different su variants.
E.g. in some su you need to use:
`su -c 'command arg12 arg2'`
and in another you must use:
`su -c command arg12 arg2`

It's also easier and more efficient to pipe a bunch of commands (e.g. a script) into a single su instead of invoking su for each of the lines. Also, piping allows to set variables and use them in the following lines etc.

### Reproduce method/复现方法

```
adb shell
echo whoami | su
echo whoami | su --no-pty
```

### Expected behavior/预期行为

output:
```
root
root
```

### Actual behaviour /实际行为

outputs:
```
whoami
root

```

### Device Name/设备名称

doesn't matter

### OS Version/系统版本

doesn't matter

### APatch Version/APatch 版本

APatch_11039-45-g60281ce_11084-release-signed.apk  -> 60281ce

### Kernel Version/内核版本

doesn't matter

### KernelPatch Version/KernelPatch 版本

doesn't matter

### Additional context/其他信息

not relevant

hg42, 2025-06-13:
it's a design problem, so logs and some other data are not relevant

hg42, 2025-06-13:
I should note that:
`echo whoami | adb shell su`
works like expected.

su only fails to execute the command, when used in Termux or ConnectBot or in another shell or invoked from an app.

Admirepowered, 2025-06-14:
Ref: https://github.com/topjohnwu/Magisk/commit/4864c1112aeb3ffb3ca27f9d9ae68a9e438960f9