bmax121/APatch #1089: 模块安装功能故障

taherenfa, 2025-06-14:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

模块安装故障

### Reproduce method/复现方法

安装任意一个模块，无其它步骤

### Expected behavior/预期行为

绝大多数模块可以成功解压并执行安装流程

### Actual behaviour /实际行为

无论什么模块，安装时均提示不支持的压缩算法，包括zip默认的算法也提示不支持的压缩算法，使用apd module install *.zip也是相同的结果

### Screenshots/截图

[image]

### Device Name/设备名称

一加ACE(PGKM10)

### OS Version/系统版本

15

### APatch Version/APatch 版本

11084

### Kernel Version/内核版本

5.10.226

### KernelPatch Version/KernelPatch 版本

0.12.0

### Additional context/其他信息

-

taherenfa, 2025-06-14:
> I think the module, you try to install is corrupt, with wrong compression.

This is indeed a bug in APatch. The zip test results from other tools are normal and can be previewed

[image]

[image]

[image]

VisionR1, 2025-06-14:
Strange.
Can you send the file?

taherenfa, 2025-06-14:
> Strange. Can you send the file?

MD5:f7447ede237f8b2a2881df55870e91ee

This bug is currently the latest CI issue, and the old version is normal. It should have been caused by the zip dependency explosion four days ago, which is similar to the problem with KernelSU

[Zygisk-Next_521.zip](https://github.com/user-attachments/files/20738441/Zygisk-Next_521.zip)

VisionR1, 2025-06-14:
So, is the same with this? 
https://github.com/Dr-TSNG/ZygiskNext/releases/tag/v1.2.9

taherenfa, 2025-06-14:
> So, is the same with this? https://github.com/Dr-TSNG/ZygiskNext/releases/tag/v1.2.9

y

VisionR1, 2025-06-14:
Yeah, i have latest APatch nightly release, and I can install normal this.

Try to download again the file, and install it.

VisionR1, 2025-06-14:
[image]
[image]

VisionR1, 2025-06-14:
In magic mount mode works

commit ac4f3507 (2025-06-17): fix #1089 downgrade zip lib