bmax121/APatch #1136: Reproducible crashes caused by installing patched boot.img with APatch Manager

Schweeeeeeeeeeeeeeee, 2025-07-20:
### Please check before submitting an issue | 在提交 Issue 前请检查

- [x] I searched the issues and didn't found anything relevant | 我已经搜索了 Issues 列表，没有发现于本问题相关内容
- [x] If the patch fails or the image cannot be booted after flashing the new boot.img, visit KernelPatch to clarify your doubts | 修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload the bug report file in APatch Manager > Settings > Send logs | 我会上传 Bug Report 文件从 APatch 管理器 > 设置 > 发送日志
- [x] I know how to reproduce the issue, which might not be specific to my device | 我知道如何重新复现这个问题

### Version requirements | 版本要求

- [x] I'm using the latest CI version of APatch Manager | 我正在使用最新 CI 版本

### Bug description | 描述 Bug

## Current Behavior
The phone experiences crashes leading to an instant black screen and automatic reboot when:
- Opening Fennec Browser and performing a search.
- Opening ReVanced Manager after 1.5-5 seconds.
- Opening Obtainium, particularly when adding an app (e.g., searching for Bitwarden or adding an app via URL like the Mullvad VPN app).

## Device Information
/codename lynx
/mods patched Apatch boot.img

### Reproduce method | 复现方法

1. **Fennec Browser**:
   - Open Fennec Browser.
   - Perform a search in the browser.
   - Observe the phone crashes with a black screen and reboots.

2. **ReVanced Manager**:
   - Open ReVanced Manager.
   - Wait 3-5 seconds.
   - Observe the phone crashes with a black screen and reboots.

3. **Obtainium**:
   - Open Obtainium.
   - Go to "Add app" and search for an app (e.g., Bitwarden) or add an app via URL (e.g., Mullvad VPN app).
   - Observe the phone crashes with a black screen and reboots.

### Expected behavior | 预期行为

Apps such as Fennec Browser, ReVanced Manager, and Obtainium should open and function without causing the system to crash or the phone to reboot.

### Actual behavior | 实际行为

Completely fresh install of lineageos worked tested all stuff no problems i try it with APatch installed crashes every time at the same locations

### Logs | 日志

[APatch_bugreport_2025-07-20_13_39.tar.gz](https://github.com/user-attachments/files/21334608/APatch_bugreport_2025-07-20_13_39.tar.gz)

### Device name | 设备名称

lynx

### OS version | 系统版本

LineageOS and LineageOS for microG

### APatch version | APatch 版本

11107

### Kernel version | 内核版本

6.1.99-android14-11-gd7dac4b14270-ab12946699

### KernelPatch version | KernelPatch 版本

0.12.0

### Other information | 其他信息

https://gitlab.com/LineageOS/issues/android/-/issues/9135
Accidentally reported it here 🥲

Schweeeeeeeeeeeeeeee, 2025-07-29:
Questions?

albotroz, 2025-08-04:
Maybe same problem?
https://github.com/bmax121/APatch/issues/879

I have Pixel 6a and can reproduce it with:

- Lineage-22.2-20250802-nightly-bluejay-signed
- APatch v11107

- Octopus (learning app)
- TinyCam
- Obtanium
- ReVanced Manager

albotroz, 2025-09-26:
Yesterday, my Pixel 6a kept restarting all day.

I hadn't changed anything beforehand.
Since I didn't have time, I turned it off.
Today, no restarts—except when I open most of the apps (see above).