bmax121/APatch #1138: Can’t open webui

s1ej1kets0, 2025-07-22:
### Please check before submitting an issue | 在提交 Issue 前请检查

- [x] I searched the issues and didn't found anything relevant | 我已经搜索了 Issues 列表，没有发现于本问题相关内容
- [x] If the patch fails or the image cannot be booted after flashing the new boot.img, visit KernelPatch to clarify your doubts | 修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload the bug report file in APatch Manager > Settings > Send logs | 我会上传 Bug Report 文件从 APatch 管理器 > 设置 > 发送日志
- [x] I know how to reproduce the issue, which might not be specific to my device | 我知道如何重新复现这个问题

### Version requirements | 版本要求

- [x] I'm using the latest CI version of APatch Manager | 我正在使用最新 CI 版本

### Bug description | 描述 Bug

在APatch管理器的模块界面点击“打开”按钮无法进入webui

### Reproduce method | 复现方法

打开APatch管理器,进入系统模块页面,找到一个自带webui的模块后点击“打开”

### Expected behavior | 预期行为

应当进入模块自带的webui

### Actual behavior | 实际行为

显示“网页无法打开
位于 https://mui.kernelsu.org/index.html 的网页无法加载，因为:net::ERR_INVALID_RESPONSE”

### Screenshots | 截图

[image]

### Logs | 日志

[APatch_bugreport_2025-07-22_17_03.tar.gz](https://github.com/user-attachments/files/21363583/APatch_bugreport_2025-07-22_17_03.tar.gz)

### Device name | 设备名称

红米K40(alioth) M2012K11AC

### OS version | 系统版本

LineageOS 21-20240710-NIGHTLY-alioth Android14

### APatch version | APatch 版本

11107

### Kernel version | 内核版本

4.19.325~RealKing//v18.0-Artful//42b5fa8e2f38

### KernelPatch version | KernelPatch 版本

0.12.0

### Other information | 其他信息

之前使用KernelSU时可以打开相同模块的webui(内核已更换,现在只有APatch)

Admirepowered, 2025-07-23:
[monitor: 😋 tracing, zygote64: 😋 injected, daemon64: 😋 running (Root: KernelSU
内核仍有kernelsu，卸载干净了再说

s1ej1kets0, 2025-07-23:
> [monitor: 😋 tracing, zygote64: 😋 injected, daemon64: 😋 running (Root: KernelSU 内核仍有kernelsu，卸载干净了再说

下载KernelSU管理器提示只支持GKI,请问这是什么原因?

Ai686Leo, 2025-07-27:
> > [monitor: 😋 tracing, zygote64: 😋 injected, daemon64: 😋 running (Root: KernelSU 内核仍有kernelsu，卸载干净了再说
> 
> 好的,我换一个内核
> 下载KernelSU管理器提示只支持GKI,请问这是什么原因?

我想这里是apatch

closed 2025-07-28 by Admirepowered