bmax121/APatch #1144: Revanced apps become unpatched after a phone reboot

eblis, 2025-07-29:
### Please check before submitting an issue | 在提交 Issue 前请检查

- [x] I searched the issues and didn't found anything relevant | 我已经搜索了 Issues 列表，没有发现于本问题相关内容
- [x] If the patch fails or the image cannot be booted after flashing the new boot.img, visit KernelPatch to clarify your doubts | 修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload the bug report file in APatch Manager > Settings > Send logs | 我会上传 Bug Report 文件从 APatch 管理器 > 设置 > 发送日志
- [x] I know how to reproduce the issue, which might not be specific to my device | 我知道如何重新复现这个问题

### Version requirements | 版本要求

- [x] I'm using the latest CI version of APatch Manager | 我正在使用最新 CI 版本

### Bug description | 描述 Bug

I'm not sure if this is by design, but feels like a problem to me. 

Every time I restart my phone all the Revanced patched apps stop working, I need to unpatch the apps and patch them again and install via mount (root).

But on next reboot they're broken again. 

I tested with YouTube, YouTube music and Reddit Sync.

### Reproduce method | 复现方法

1. Patch YouTube music with Revanced
2. Install via mount (root)
3. Restart phone
4. YouTube music says it's patched, but it's not. 
5. Need to unpatch the app and start from 1. again

### Expected behavior | 预期行为

Phone restart should not break revanced apps

### Actual behavior | 实际行为

Phone restart breaks revanced apps

### Device name | 设备名称

Realme GT neo 

### OS version | 系统版本

Android 14

### APatch version | APatch 版本

11107

### Kernel version | 内核版本

4.19.321

### KernelPatch version | KernelPatch 版本

0.12.0

### Other information | 其他信息

Nothing

RevontuletCXVII, 2025-09-13:
This issue was [marked as closed in August](https://github.com/bmax121/APatch/issues/923) as "fixed", but still occurs in the latest nightly (11119). 

This is a pretty major problem with how APatch operates, as it means there's a gap between initial boot and activation of the user space where APatch isn't active and cannot accept or deny SU requests and likely stays in a state vulnerable to malicious action.

306bobby-android, 2025-09-25:
> This issue was [marked as closed in August](https://github.com/bmax121/APatch/issues/923) as "fixed", but still occurs in the latest nightly (11119).
>
> This is a pretty major problem with how APatch [...]

It seems to be less that, and more of an issue of how the android shell is able to mount with APatch 

For example: if YouTube is unmounted using magisk or ksu, you can run the service.d script and remount the app successfully. On APatch, the app will not mount via the script, no matter how I've ran it

nxgensoftware, 2025-11-11:
Can confirm I'm also affected by the issue. Xiaomi Redmi Note 12 Pro running Hyperos EU 2.0.5.0. Tried killing the app without tasker by doing /system/bin/am com.google.android.youtube as su on termux to no avail.

MeGaNeKoS, 2025-12-14:
Hello, I open another issue ticket as I feel it more appropiate. But in short, there's a weird behavior when you execute pm with having output rediraction and not.
```sh
  v_plain="$(pm path "$pkg")"
  echo "[INFO] $(date '+%F %T') plain='$v_plain'"
  v_nullerr="$(pm path "$pkg" 2>/dev/null)"
  echo "[INFO] $(date '+%F %T') stderr_to_devnull='$v_nullerr'"
```
In this case, the first approach will always return `cmd: Failure calling service package: Failed transaction (2147483646)` while the second approach will return `package:/data/app/~~arY4yjg3P9xYKDk3T6Wn8w==/com.google.android.youtube-PgB8vwNmEZ0ztAgGJhCIlw==/base.apk`.

I've try open PR on revanced manager to help support this case, but as their response might not design their apps on broken fundamen. And, I see someone on their discord linking to this ticket. But it would be great to had a better insight why this happens on APatch as I feels this our problem.

For other, you can do this quick fix. Open the revanced boot shell script, and change
`stock_path=$(pm path com.google.android.youtube | grep base | sed "s/package://g" )`
to
`stock_path=$(pm path com.google.android.youtube 2>$1 | grep base | sed "s/package://g" )`

https://github.com/bmax121/APatch/issues/1324