bmax121/APatch #1146: 添加对Chrome OS boot镜像的支持

wwerqsj4678, 2025-08-02:
### Is your request related to a specific issue? | 你的请求是否与某个问题相关？

有些设备拥有Chrome OS的分区方案（例如Pixel C），导致内核无法被修补

[image]
[image]

### Describe the solution you'd like | 描述你想要的解决方案

添加像magisk一样的Chrome OS boot 镜像的支持

### Describe the alternatives you've considered | 描述您考虑过的备选方案

No

Admirepowered, 2025-08-11:
APatch不修补ramdisk，Chrome OS分区是什么，有boot镜像吗

wwerqsj4678, 2025-08-17:
Chrome OS拥有boot镜像，但是分区结构不同

wwerqsj4678, 2025-08-17:
可参考此issue
https://github.com/bmax121/KernelPatch/issues/194
boot.img在下面的链接（提取自/dev/block/mmcblk0p1）
https://wwdu.lanzoup.com/ilcOr33qti3i 密码a7b8